yamcs/yamcs #841: assigned roles not properly visible on yamcs web?

duane-impulse, 2023-10-31:
I am combining OpenIDC with YamlAuth. OIDC authenticates, and then the Yaml Auth provides permissions. This is done by assigning some usernames specific roles (which are defined in roles.yaml) in users.yaml.

It appeared the roles weren't taking and I spent some time trying to figure out where I was going wrong. But I think I have finally deduced that it actually is _mostly_ working, just that the roles aren't properly visible on the yamcs web interface of the users? It does appear permissions are as expected. 

However, if I define one of the roles as 'default', that role name does appear on the yamcs web interface (even if a user should have been given a different role and appears to have permissions as such).

Upon a further glance, the openmct-yamcs plugin actually does appear to display the roles I would expect.

fqqb, 2023-10-31:
In the user profile, you see only 'internal' roles. That is: roles assigned through the Admin UI (and stored in Yamcs DB).
When using an 'external' authorization mechanism, like YamlAuthModule, only the privileges are imported.

The 'default role' capability is applied regardless of YamlAuthModule.

So your description sounds within expectation, though certainly there is some confusing aspects that could be improved.

duane-impulse, 2023-10-31:
Thanks for the response! It is good to know now that it is working, even if the exact behavior isn't the most intuitive.

I do think I'd come across references to 'internal' vs 'external' auth in the docs but had not connected the dots that _YamlAuthModule_ is 'external' (although this certainly makes sense in some contexts), because _users.yaml_ felt just as 'native' as  _roles.yaml_.

Thanks again!