yamcs/yamcs #843: Facing issues in authenticating with YAMCS websockets in browser using Javascript

bangpradyumna, 2023-11-06:
Hello,
I am trying to connect to YAMCS websocket from a javascript based webapp but turns out the popular WebSocket library doesn't support sending Authentication headers during the connection. 

Please suggest some ways to authenticate with YAMCS websockets using javascript running on the browser. I tried reading the code for webapp-sdk but couldn't really find WebsocketClient sending the access_token at all during connection request. Am I missing something ?

I'd really appreciate any help in this regard,

Thank you
Pradyumna

fqqb, 2023-11-06:
That's a browser limitation. yamcs-web works around this by storing the access token to a cookie, which then gets send on the websocket request.

The code on the server-side which handles this is here: https://github.com/yamcs/yamcs/blob/ebcdc82a2bb85d4945a97673626b0a42a30f69c8/yamcs-core/src/main/java/org/yamcs/http/HttpHandler.java#L106

That should be improved at some point, and there's a few ways to do that, but please see if this is working for you.

bangpradyumna, 2023-11-16:
> That's a browser limitation. yamcs-web works around this by storing the access token to a cookie, which then gets send on the websocket request.
>
> The code on the server-side which handles this is [...]

Thanks alot @fqqb 
Unfortunately, Its not possible to use the cookie method since we don't plan on hosting the frontend and yamcs on the same domain. It looks like you can't send cookies if the domains are different (please correct me if I'm wrong).

Would it be possible to elaborate on the other ways of doing this that you mentioned?

Thanks again

fqqb, 2023-11-20:
For cross-domain, or really any advanced requirements, I'd look into handling authentication outside of Yamcs, like with Keycloak (via [OpenIDAuthModule](https://docs.yamcs.org/yamcs-server-manual/security/authmodules/openid/)) or Apache/nginx (via [RemoteUserAuthModule](https://docs.yamcs.org/yamcs-server-manual/security/authmodules/remote-user/)).

A pure-Yamcs non-cookie browser solution for WebSocket would involve some new development.