yamcs/yamcs #844: Specifying a StringDataEncoding with a Variable -> DynamicValue size specification isn't possible

Niautanor, 2023-11-06:
## The problem

The following StringDataEncoding the the XTCE MDB

```
               <StringDataEncoding>
                  <Variable maxSizeInBits="4528">
                     <DynamicValue>
                        <ParameterInstanceRef parameterRef="Message_Length" />
                        <LinearAdjustment slope="8" intercept="-8" />
                     </DynamicValue>
                  </Variable>
               <!--<SizeInBits><Fixed><FixedValue>256</FixedValue></Fixed></SizeInBits>-->
               </StringDataEncoding>
```

causes an error during startup that reads `Message: SizeInBitsor Variable not specified for the StringDataEncoding`

This is from https://github.com/yamcs/yamcs/blob/ebcdc82a2bb85d4945a97673626b0a42a30f69c8/yamcs-xtce/src/main/java/org/yamcs/xtce/xml/XtceStaxReader.java#L1240

This happens because `readVariableStringSize()` in the `DynamicValue` case does not set the SizeType (but it does `setDynamicBufferSize()` which is the dynamic equivalent to `setSizeInBits()`).

When the `<SizeInBits>` element is specified additionally (i.e. not commented out), everything works (i.e. the size of the string is dynamically computed).

This happens because the presence of a dynamic size specification is checked first during extraction (and the fixed size specification is ignored in this case).

https://github.com/yamcs/yamcs/blob/ebcdc82a2bb85d4945a97673626b0a42a30f69c8/yamcs-core/src/main/java/org/yamcs/mdb/DataEncodingDecoder.java#L170

## Potential solutions

1. `readVariableStringSize()` could also set the SizeType to `FIXED`. The interpretation of this would be that
   - The memory area that the string element occupies is specified by either `sde.getSizeInBits()` or `sde.getDynamicBufferSize()`
   - The string is extracted from this memory area according to SizeType where `FIXED` means "the string is the entire memory area"
2. A new `DYNAMIC` SizeType could be added, set in `readVariableStringSize()` and added to all switch statements that examine the SizeType.

I personally like the decoupling of memory area and actual string part of the memory area of the first solution (it would also be a whole lot easier to implement).

Additionally, the check should probably be modified to check that either `getSizeInBits()` or `getDynamicBufferSize()` (and not both) are set in addition to SizeType. (and a space should be added in front of the or)

(I can make a pull request for this but I wanted to get some feedback on the potential solutions first)

xpromache, 2023-11-07:
Hello,

Thanks for reporting this. Your solution 1 makes sense.  However I do not understand why this file:
https://github.com/yamcs/yamcs/blob/ebcdc82a2bb85d4945a97673626b0a42a30f69c8/yamcs-core/src/test/resources/xtce/strings-tm.xml#L56

used in this test:
https://github.com/yamcs/yamcs/blob/ebcdc82a2bb85d4945a97673626b0a42a30f69c8/yamcs-core/src/test/java/org/yamcs/mdb/XtceStringDecodingTest.java

does not exhibit the problem you report.

Probably I'm missing something. So maybe you can make another test case (same xml, same java files) together with your proposed fix.

Also if you don't mind signing the CLA...
https://yamcs.org/static/Yamcs_Contributor_Agreement_v2.0.pdf

Thanks!
nicolae

Niautanor, 2023-11-10:
Hi Nicolae,

I have pushed a branch in my fork to add a test case to the file you linked (and my proposed fix in a separate commit). The reason why the linked test case works is because the string data encoding has a `LeadingSize` element which sets the `SizeType` to `LEADING_SIZE`. YAMCS extends the XTCE spec by allowing string data encodings with neither a DynamicValue or a Fixed size specification (or alternatively a DynamicValue with a parameter reference to `_yamcs_ignore`) which will set the `div` to `null` and the `sizeInBits` to 0 and interpret the buffer size to be the size of the string (as determined by the prefixed size).

I have created a merge request from this (#846)

Regarding the CLA, where should I send it to after I signed it?

xpromache, 2023-11-10:
Thank you very much for your contribution. I will check it one of the following days.

You can send the CLA to yamcs@spaceapplications.com

Niautanor, 2024-06-24:
I am going to close this since the related pull request was merged.

closed 2024-06-24 by Niautanor