yamcs/yamcs #845: Two bugs related to application.properties

0lionelzhang0, 2023-11-06:
### application.properties does not seem to support booleans
provideJvmVariables: bool (.yaml file)
bool = true (application.properties)

``` 09:40:57.165 _global [1] YamcsServer yamcs.myproject.yaml: services[5].args: provideJvmVariables is of type STRING, but should be BOOLEAN instead ```

### default value doesn't seem to work
yamcs-web
- tag: ${foo:bar}

- when neither foo or bar are defined in application.properties, the tag is "bar" (expected)
- when foo is defined as "barfoo", the tag is "barfoo" (expected)
- when foo is not defined and bar is defined as "foobar", the tag is "bar", when should be "foobar" (bug?)

fqqb, 2023-11-07:
For the first issue, I think you're just missing the variable syntax: `provideJvmVariables: ${bool}`

For the second issue, indeed it should become possible to have spaces inside the default value. That's a bug.

Semantically, I think a nested variable could be expressed like this,`${foo:${bar}}`. However, this will not work currently either.

0lionelzhang0, 2023-11-07:
Sorry about that first one! My mistake...

Thanks for looking into the second one, that would be helpful to have!

fqqb, 2024-01-02:
Nesting issue addressed in PR #858

fqqb, 2024-03-06:
Should all be good as of v5.9.x

closed 2024-03-06 by fqqb