yamcs/yamcs #855: Serving parameter subscriptions with large array parameters requires very large writeBufferWaterMark

merose, 2023-12-14:
It appears that all parameters must be converted to protobuf or JSON in-memory before they can be sent as updates for subscriptions. This can cause very large writeBufferWaterMark to be needed for large array parameters, as for spectrometer spectra, e.g. It would be highly desirable to stream these values instead of constructing them in-memory, to reduce per-client memory requirements in Yamcs.

xpromache, 2024-02-23:
Marc, are those clients OpenMCT? 
Is it possible to switch to binary protobuf encoding instead of json? That will certainly save some memory.

I'm attempting now some tricks to save memory on the server by avoiding conversion to protobuf. 
But fundamentally the problem of the writeBufferWaterMark cannot be avoided so easily: assuming  you have  a big chunk of data to be transferred to the client, and it cannot be transferred instantly you either have to hold further data reception/processing  until the client got the current data or to put it somewhere for it to be transferred at the slow speed of the client.  That somewhere is essentially the netty buffer with the high/low watermarks - it has to have at least the capacity of the largest data chunk produced "at once".

What we can try to do is reduce the size of the data by encoding it into binary not json.

One other thing to consider is that if there are multiple clients and one large parameter value, that gets duplicated for each client - the reason is that each client has its own subscription and the data consist of a combination of parameter values custom for that client.
For example client 1 subscribes to parameters A and B, client 2 subscribes to parameters A and C, so we create two deliveries (A,B) and (A,C) - ultimately when each delivery gets converted to binary to be sent on the socket, the parameter A data gets duplicated.

I'm thinking to implement some sort of large data service - practically when encountering a large value, we put it in a buffer somewhere and send only the reference to that buffer not the value itself. The buffer would be shared between all clients, have a maximum size and not counted against the client water marks. The client would be a bit more complicated but hopefully not too much.

The large data service requires more time to implement so I will try first a trick that eliminates the conversion of  pure java Value into protobuf Value.

akhenry, 2024-03-07:
@xpromache The use case for large buffers is for large binary arrays, which causes any client (including the Yamcs Web UI) to be dropped due to the buffer immediately overflowing during message serialization. 

We get around this by configuring arbitrarily large buffers. One consequence of this is that any slow client can eventually bring down Yamcs through "slow loris" type behavior. By configuring these large send buffers across the board we are effectively disabling an important fail-safe against this behavior.

A possible solution here is being able to specify write buffer sizing per-websocket, rather than application-wide. Perhaps clients could request a buffer that is smaller than the server configuration in the initial websocket request? That way we can isolate the small number of subscriptions that need large buffers, and allow the majority of subscriptions to benefit from the slow-client protection that small buffers offers (by dropping clients when the buffer fills up).

akhenry, 2024-03-07:
> Is it possible to switch to binary protobuf encoding instead of json? That will certainly save some memory.

Open MCT did experiment with protobuf because of the server-side benefits, but we saw an increase in memory usage of 200-300% in the client. We ran out of time to do a deep investigation into what was going on, and we may come back to it in the future, because of the server-side benefits.