yamcs/yamcs #869: Security login regression in Yamcs v5.8.8

fgretief, 2024-01-31:
During the Yamcs training session yesterday there was an excersize that enabled secruity. Today, I was trying this out on my own machine with a fresh copy of the Yamcs repository.  I checked out the 'master' branch and added these two files to the `examples/simulation/src/main/yamcs` folder.

`etc/security.yaml`:
```yaml
enabled: true
authModules:
  - class: org.yamcs.security.YamlAuthModule
```

`etc/users.yaml`:
```yaml
bruce:
    password: wayne
    superuser: yes
```

Then I build Yamcs as described in the README.md file. Once finished I started the example with: 
```
./run-example.sh simulation
```

Then I opened my browser and connected to the Yamcs server and the login paged displayed as expected. I typed in the username and password and clicked the `Sign in` button. 

But instead of going to the main Yamcs page, it reloaded the login page. And it keeps going back to the login page no matter what user I use. The log output on the terminal looks like this:
```
15:51:35.462 _global [40] HttpRequestHandler e58532d5 GET / 200
15:51:36.992 _global [41] HttpRequestHandler c3f36e6a GET /auth/authorize?client_id=yamcs-web&state=Lw&response_mode=query&response_type=code&scope=openid&redirect_uri=http%3A%2F%2F172.18.226.11%3A8090%2Fcb 200
15:51:37.893 _global [41] HttpRequestHandler c3f36e6a GET /apple-touch-icon.png 200
15:51:37.893 _global [42] HttpRequestHandler 79c9e30c GET /favicon-16x16.png 200
15:51:50.969 _global [43] SecurityStore Successfully logged in [name=bruce, id=7]
15:51:51.015 _global [43] AuthHandler Redirecting to http://172.18.226.11:8090/cb?code=fngvXj2MlifAlw&state=Lw
15:51:51.015 _global [43] HttpRequestHandler e58532d5 POST /auth/actions/login 302
15:51:51.151 _global [40] HttpRequestHandler e58532d5 GET /cb?code=fngvXj2MlifAlw&state=Lw 200
15:51:52.036 _global [40] AuthHandler Access token request using grant_type 'authorization_code'
15:51:52.150 _global [40] HttpRequestHandler e58532d5 POST /auth/token 200
15:51:52.261 _global [40] HttpRequestHandler e58532d5 GET /auth/authorize?client_id=yamcs-web&state=Lw&response_mode=query&response_type=code&scope=openid&redirect_uri=http%3A%2F%2F172.18.226.11%3A8090%2Fcb 200
```

Later I started checking earlier commits and eventually found this commit to be the point were the regression starts: 6c709563094a6a37bd7009f4396aaab86a982983 

This is what the logs looks like for a successful login:
```
16:07:47.520 _global [40] HttpRequestHandler 58ee5836 GET / 200
16:07:48.466 _global [40] HttpRequestHandler 58ee5836 GET /auth/authorize?client_id=yamcs-web&state=Lw&response_mode=query&response_type=code&scope=openid&redirect_uri=http%3A%2F%2F172.18.226.11%3A8090%2Fcb 200
16:07:48.839 _global [40] HttpRequestHandler 58ee5836 GET /auth/assets/auth.css 200
16:07:48.849 _global [41] HttpRequestHandler e75e69ca GET /auth/assets/yamcs300.png 200
16:07:56.558 _global [43] SecurityStore Successfully logged in [name=bruce, id=7]
16:07:56.601 _global [43] AuthHandler Redirecting to http://172.18.226.11:8090/cb?code=wTRTi3fz9jDJUw&state=Lw
16:07:56.601 _global [43] HttpRequestHandler cc099398 POST /auth/actions/login 302
16:07:56.621 _global [42] HttpRequestHandler cc099398 GET /cb?code=wTRTi3fz9jDJUw&state=Lw 200
16:07:57.334 _global [42] AuthHandler Access token request using grant_type 'authorization_code'
16:07:57.456 _global [42] HttpRequestHandler cc099398 POST /auth/token 200
16:07:57.504 _global [42] HttpRequestHandler cc099398 GET /favicon.ico 200
16:07:57.823 _global [47] WebSocketFrameHandler 9016076c GET /api/websocket 101 [subprotocol: json]
16:07:58.020 _global [52] HttpRequestHandler 751fd89b GET /api/instances 200
```

Hope that helps.

Regards,
  Francois

fqqb, 2024-01-31:
Fundamentally it's not 'secure' to do auth over http. With that commit, i believe only an address like localhost or an https address would be possible.

fgretief, 2024-01-31:
> The generated redirect is to http://172.18.226.11:8090 
> Is that also the address you're using to access the website?

Yes, that is correct. I was using it under WSL2 on Windows, so that address is the Linux VM's address on the Windows machine.

> Fundamentally it's not 'secure' to do auth over http. With that commit, i believe only an address like localhost or an https address would be possible.

I do understand that it is not 'secure'. This is how it was described in the training exercise from the Yamcs Exercise Booklet (page 6).  During Step 4 (Verification) we used the the same `http://ipaddress:8090` address type to access the Linux VMs provided to us. I was just repeating the exercise.

I was just surprised that it did not work with Yamcs v5.8.8. With v5.8.7 it did work correctly.

fqqb, 2024-03-06:
v5.9.x introduces a `cookie` configuration option as part of the `yamcs-web` configuration section.
See https://docs.yamcs.org/yamcs-server-manual/web-interface/configuration/

The defaults are less strict, such that the issue here described, would not appear. Then it is assumed that these properties are configured where needed, as part of server hardening.

closed 2024-03-06 by fqqb