yamcs/yamcs #870: Misleading "Alarm Info" in MDB?

0lionelzhang0, 2024-02-01:
Yamcs alarms (version: 5.8.8) by default follow the XTCE rangeForm="outside" convention, which is working as expected. However, the range information displayed in MDB appears to follow the rangeForm="inside" convention which can be misleading.

For example, the following alarm triggers on values from -inf to 4 
``` 
<DefaultAlarm>
  <StaticAlarmRanges>
    <SevereRange minExclusive="4" />
  </StaticAlarmRanges>
</DefaultAlarm>
```

However, this is what is displayed in the MDB UI. I think it would make more sense if this showed (-inf, 4) instead.
[image]

Similarly, when both min and max are defined:
```
<DefaultAlarm>
  <StaticAlarmRanges>
    <SevereRange minExclusive="2" maxExclusive="4" />
  </StaticAlarmRanges>
</DefaultAlarm>
```

The following is shown in the web UI. But I believe it would make more sense to show (-inf, 2), (4, +inf) instead to reflect the actual trigger ranges.
[image]

fqqb, 2024-02-02:
indeed, that'd be better

fqqb, 2024-03-06:
Addressed in 5.9.x