yamcs/yamcs #886: `setParameterValue` does not set a rawValue for enums

davetsay, 2024-03-26:
Even though passing the raw value to the http api, yamcs converts to the label value. Similarly, if I use the `SET VALUE` button in the yamcs UI, the raw value is `-`. This results in calls to the `samples` endpoint with `&useRawValue=true` to fail to return results, even though they are present in yamcs `HISTORICAL DATA` view.

Steps to reproduce
1. modify quickstart so that one of the test enums uses `local` datasource.
2. run yamcs quickstart.
3. search for the modified parameter and use the `SET VALUE` button.
4. observe no raw value is set.
5. `setParameterValue` http api call will provide similar results to `SET VALUE`.

body of `setParameterValue` call
```
{
  type: 'SINT64',
  sint64Value: 0,
}
```

Enum XTCE
```
<EnumeratedParameterType name="EnumerationTest">
  <UnitSet />
  <IntegerDataEncoding sizeInBits="3" />
  <EnumerationList>
    <Enumeration value="0" label="ENUM_VALUE_0" />
    <Enumeration value="1" label="ENUM_VALUE_1" />
    <Enumeration value="2" label="ENUM_VALUE_2" />
    <Enumeration value="3" label="ENUM_VALUE_3" />
    <Enumeration value="4" label="ENUM_VALUE_4" />
    <Enumeration value="5" label="ENUM_VALUE_5" />
    <Enumeration value="6" label="ENUM_VALUE_6" />
    <Enumeration value="7" label="ENUM_VALUE_7" />
  </EnumerationList>
</EnumeratedParameterType>

<Parameter name="Enum_Para_1" parameterTypeRef="EnumerationTest">
  <ParameterProperties dataSource="local"/>
</Parameter>
```

fqqb, 2024-03-28:
It's because the numeric value of an enumeration is not actually a raw value.
You are setting the engineering value, just that the engineering value has two identities (both a value and a label).

This explains why the ``useRawValue`` option doesn't work.

So it seems we should add another option to the endpoint, to perform the sampling on the numeric value of an enumeration state.

merose, 2024-03-28:
I'm not sure exactly what's needed. This is a request from OpenMCT. Perhaps
they can respond and give you more info.

fqqb, 2024-03-29:
When you set a local parameter, you're setting the engineering value.

In your example, `0` is the numeric form of the engineering value. There is no raw value, hence why useRawValue does not work.

```
{
  type: 'SINT64',
  sint64Value: 0
}
```

What I was suggesting (assuming you do want to sample local enumeration parameters), is that we add another option to this API endpoint, to perform the sampling based on the numeric form of the engineering value.

akhenry, 2024-03-29:
@fqqb I see, thanks for the explanation. Yes an additional parameter here would fix the issue on our side, thank you.

merose, 2024-03-29:
It looks like the Get Parameter Value processing API and the Sample
Parameter Values archive API aren't quite uniform in their return values
for enumerated types.

Processing, Get Parameter Value
type=ENUMERATED
Both sint64Value and stringValue are populated

Archive, Sample Parameter Values
type=STRING
only stringValue is populated

I think making Sample ParameterValues return the same structure as Get
Parameter Value would solve the issue.

davetsay, 2024-05-29:
@fqqb , there was some back and forth discussion and I just wanted to circle back and make sure were on the same page. I think @merose 's suggestion makes the most sense. Would you agree?

fqqb, 2024-06-19:
I had looked at it a while back, but it turns out there's no 'quick fix' because it appears the Parameter Archive is only returning the stringValue. So that's something for @xpromache to look at first.

On the last question, since the goal of the samples call is to calculate numeric aggregates (min, avg, max) over an arbitrary period, this does lead to the question what the string value should be for an averaged numeric state (or even what the numeric value should be). For example an enumeration:

* 1: AA
* 2: BB

Could theoretically lead to sample values that are unmatched:
* (n=1, min=1/AA, avg=1/AA, max=1/AA): OK
* (n=2, min=1/AA, avg=1.5/??, max=2/BB): Questionable 

I think more appropriate for discrete values would be the Parameter Ranges call, which was designed explicitly for this purpose: https://docs.yamcs.org/yamcs-http-api/parameter-archive/get-parameter-ranges/. With the correct arguments, it is capable of returning small responses, even over a large time period. I've been wanting to integrate an example use of that inside the Yamcs Web UI (via a new visualization activated only for non-numeric parameter types).