yamcs/yamcs #895: ConcurrentModificationException is possible during CFDP options in DataFile class

merose, 2024-05-15:
The DataFile class has an unsynchronized list "dataFileSegments" that can be read and modified by different threads. The particular problem we found was caused by a ConcurrentModificationException in getReceivedSize(). Apparently another thread added a segment using addSegment() at the same time.

The easiest fix I see is to make all accesses of "dataFileSegments" synchronized, probably by modifying the instantiation to use `Collections.synchronizedList(new ArrayList<>())`.

merose, 2024-05-15:
Submited a PR: https://github.com/yamcs/yamcs/pull/896

xpromache, 2024-05-20:
I set all methods as synchronized instead because adding a segment may involve multiple operations on that list (for example merging two segments when the gap between them is filled is two operations: 1. extend the first segment, 2. remove the second segment) and in that case the getReceivedSize may result in a size larger than the received size.

commit 7ef4a273 (2024-05-20): Synchronize all methods of DataFile

CFDP operations are all from one thread with one exception: the getReceivedSize which is called via the API. That means the method can be called when an incoming segmnet is being added to the list causing ConcurrentModificationException.
Fixes #895
closed 2024-05-20 by xpromache