yamcs/yamcs #900: Incorrect calculation of data offset for TM Frames with secondary header

pmccauley1, 2024-06-11:
I think there might be a bug in the calculation of the `dataOffset` field in `TmFrameDeocder`.  The CCSDS specification states the value of the secondary header length is "the total length of the Transfer Frame Secondary Header in octets **minus one**.  I think the line below should look like this:

`dataOffset += secHeaderLength + 1;    // "+ 1" to account for the length being zero-based per CCSDS standard`

See link here: https://github.com/yamcs/yamcs/blob/76be9c237fa25d471dc5ab86e3b5db6c1688ffcb/yamcs-core/src/main/java/org/yamcs/tctm/ccsds/TmFrameDecoder.java#L86

xpromache, 2024-06-12:
Thanks for the report.
I will set it already two lines before:

` int secHeaderLength = 1 + data[dataOffset] & 0x3F;`

do you agree?

Can you share some binary frames I can use in unit tests?

pmccauley1, 2024-06-12:
I think that should be fine?  As long as `secHeaderLength` is assuming the
actual length and not the value represented in the header (one less than
the length), that should be good.  Regarding frames, let me check and get
back to you.

commit c5247e7c (2024-06-19): fixed the computation of the secondary header length

fixes #900
closed 2024-06-19 by xpromache