yamcs/yamcs #903: WriteRows InvalidProtocolBufferException

altsv, 2024-06-20:
Hello,

I am running into an error code 500 `InvalidProtocolBufferException` when attempting to write rows onto the tm table.
Currently I am experimenting with using the HTTP API to write in packets. To test this, I created the following script:

**write_rows.py**
```
import requests

def post(url, payload):
    response = requests.post(url, json=payload)

    if response.status_code == 200:
        try:
            print(response.json())
        except Exception:
            print("FAILED TO CONVERT TO JSON")
            print(response.text)
    else:
        print(f"Failed to execute SQL: {response.status_code}, {response.text}")

# This is from a readRows call
yamcsRowFromRead = {
    "columns": [
        {"id": 0, "name": "gentime", "type": "TIMESTAMP"},
        {"id": 1, "name": "seqNum", "type": "INT"},
        {"id": 2, "name": "rectime", "type": "TIMESTAMP"},
        {"id": 3, "name": "status", "type": "INT"},
        {"id": 4, "name": "packet", "type": "BINARY"},
        {"id": 5, "name": "ertime", "type": "HRES_TIMESTAMP"},
        {"id": 6, "name": "link", "type": "ENUM"},
        {"id": 7, "name": "pname", "type": "ENUM"},
    ],
    "cells": [
        {"columnId": 0, "data": "AAABkDbmuDc="},
        {"columnId": 1, "data": "CAHADw=="},
        {"columnId": 2, "data": "AAABkDbmuDk="},
        {"columnId": 3, "data": "AAAAAA=="},
        {
            "columnId": 4,
            "data": "AAAATggBwA8AR1OfNtWBZQAAACE+szMzwnIzM0JVMzNHHD8AQzQAAEEdBiW6Y/c2QFkm6UBZJuk8LwOON7OEJkBY3tM10AmYwqAAAEM0AACiEA==",
        },
        {"columnId": 5, "data": "AAABkDbmuDkAAAAA"},
        {"columnId": 6, "data": "AAt0bV9yZWFsdGltZQ=="},
        {"columnId": 7, "data": "ABkvWVNTL1NJTVVMQVRPUi9GbGlnaHREYXRh"},
    ],
}

post(
    "http://localhost:8090/api/archive/simulator/tables/tm:writeRows", yamcsRowFromRead
)
```

This script attempts to write a single row to the tm table. The data is copy and pasted from a readRows call. The readRows response types correspond with the request body types of writeRows (ColumnInfo and Cell), which is why I opted to take the data from readRows.

Since this JSON is from readRows, my request body should be valid, but instead I am getting the following error:

Error Code 500:
```
com.google.protobuf.InvalidProtocolBufferException: Cannot find type for url: 
/yamcs.protobuf.table.WriteRowsExceptionDetail
```

This error is from TableApi.java, but I am unfamiliar with protobuf so I'm unsure how to proceed.
Is there something wrong with my POST call? Or is there something else I missed?

Thank you!

fqqb, 2024-06-21:
The error is not the best, but is hiding another exception:

```
org.yamcs.http.BadRequestException: Protocol message tag had invalid wire type.
   at org.yamcs.http.StreamingClientHandler.exceptionCaught(StreamingClientHandler.java:66)  
```

Really, what's happening here is that streaming-uploads only support protobuf input, not JSON. You may be able to get it working by adapting from the logic in ArchiveClient load_table / dump_table methods in the yamcs-client Python package.

Though, more conventional would be to use a FilePollingTmDataLink with your custom preprocessor.

altsv, 2024-07-02:
@fqqb Thank you for clearing that up! Will opt to use FilePollingTmDataLink instead for now 👍

closed 2024-07-02 by altsv