yamcs/yamcs #904: How to read raw packets which are CCSDS Space Packet Protocol format

haisamido, 2024-06-25:
How to read raw packets which are CCSDS Space Packet Protocol format?

We have telemetry files which only contain packets in the "raw" CCSDS Space Packet Protocol format, meaning there are no frames or encapsulations. How can they be imported into Yamcs?

xpromache, 2024-06-26:
You can use a file polling link:
https://docs.yamcs.org/yamcs-server-manual/links/file-polling-tm-data-link/

and then copy the files to the directory configured in the link. Yamcs will ingest them. Note that the CCSDS Space Packets do not include a timestamp by default.  Yamcs needs a timestamp when injecting a packet.
Each project may define s a secondary header which oftentimes include that timestamp. If not, you can just use the local injection time (which has the disadvantage that injecting again the same data results in duplicates).

The function of extracting the time (and the sequence count) from the incoming packet is what the pre-processor does in Yamcs. You will likely need one for your usecase.

haisamido, 2024-06-26:
Nearly all, if not all, of the CCSDS packets in question have timestamps in seconds and sub-seconds starting on the 6th byte, relative to a specific reference date, and relative to specific time scale, TAI, UTC, GPS.... The timestamps and references are specific to each APID (APPLICATION PROCESS IDENTIFIER). _Does that metadata need to be passed to the yamcs' ingest step, or is that only needed at the extraction time?_

_By ingest, is yamcs only capturing the metadata, an index of sorts, for each packet and placing that in the database or is the actual content of the packet is placed in the database?_ The latter would be harder to maintain since the database could be in the terabytes and beyond.

xpromache, 2024-06-27:
Yamcs needs indeed the timestamp to save the packet in the database. The whole packet is saved, not only the metaadata. 

We have  time decoders supporting different timescales but you cannot configure the them  based on APID - all packets from one file need to use the same timescale. If you would have different files for the different time encodings then you could have different links with the different configurations.

Otherwise you would need to implement that in a custom  pre-processor.

I would be curious to test the terabytes of data with Rocksdb. We have an option to partition per time, so you can have one database per year or per month or even per day. I imagine you will also want to partition per mission (for this we have the concept of "instance" in Yamcs).

Facebook claims to store petabytes of data in RocksDB so I think it should work. Not sure what your expectations are once the data is there, what do you want to do with it?

haisamido, 2024-07-12:
Storing the metadata, which contains the URI of the data itself, is a more scaleable solution, than storing the actual data in the database. Facebook probably uses an analogous model to the approach I mentioned. The URI can be s3://, sftp://, nfs://, etc, with the relevant mission, APID, and date/time partitioning, etc. With this approach maintainability of the database would be less complex.

**Can yamcs be used to use metadata instead of the actual data, e.g. s3:// as the source instead of being in the database?**

e.g. the URI can be something like:

s3://telemetry/ops/my_mission/data/2024/135/my_mission_2024-135-16:09:48:00_2024-135-16-16:08:33.ccsds.gz

xpromache, 2024-07-12:
No, Yamcs currently cannot store the content of the packets outside the database.

In the missions I have seen, the maximum size of the packet is somewhere at 1000-1500 bytes but many packets are much smaller, around 100-200 bytes or even smaller. I think that is a bit too small for an object store like S3.

There is also a question of read latency - retrieving from S3 takes 10-15 millisec/object, that would mean a replay would run at max 60-100 packets/second which is much less than what Yamcs can currently do (I think it can do 20-30K packets/second without problems).
 
What is trendy nowadays is indeed to store data in S3 or other external object stores but not at that small granularity. For example InfluxDB v3 stores parquet files in an object store - but the files are in the range of tens if not hundreds of megabytes.

Storing Yamcs packets into S3 at higher granularity (for example 5 minutes of consecutive packets into one S3 object) would be an interesting path to pursue.

haisamido, 2024-07-15:
You are right, storing individual APID packets would not be well suited for s3. The granularity we've opted to use is several hours worth of packets, but is configurable, by mission, APID, etc.. You're right latency could be of concern, however, that's is easily mitigated with caching. That being said s3 keeps on getting faster. Had Yamcs been able to use pre-existing stored APID packets, it would make our hypothetical and notional migration to Yamcs more viable :-)

xpromache, 2024-07-15:
I hadn't noticed that in your s3 URL example you used a time range, so yes, I agree that storing a number of consecutive packets in one object makes sense.
The per APID bit will cause additional difficulties - when we do a replay we have to play packets in order (no matter what the APID) so that would mean merging multiple APID files to get a consistent stream. 
But it is doable - Yamcs already implements partitioning which could be configured to work per APID.

In addition the storage engine used by Yamcs is plugable so in theory it is possible to implement a storage backed by S3 like you suggest although it does constitute a serious effort. 

My immediate plans for the Yamcs storage is to optionally move away the storage from the embedded RocksDB to a remote RocksDB. That will make it more friendly to use in containerized environments such as Kubernetes.
I still want to keep the option of running everything embedded because I consider that as a plus for Yamcs that it runs on everyday's developer laptop. 

Your proposal to keep a RocksDB "local" for indexing and use a object store for data makes sense but it is not something I would do in my spare time in the immediate future.

Anyway, the discussion seems to have moved away from the subject so I guess we can close it here, or if you want you can keep it open but change the subject.

haisamido, 2024-07-15:
Thank you for the reply. I'll change the topic. What do you recommend I change it to?

xpromache, 2024-07-15:
I changed the topic myself. 

Thinking more about it: to do a complete backend on top of S3 is quite some work due to the writable nature of the Yamcs archive (we would need to insert packets in the middle of sorted file and that concurrently from multiple threads - that's basically what RocksDB does).

If instead we would focus only on read, things would be much simpler:
- we store realtime and newish data into RocksDB as we do now 
- move data older than n days to S3.

The data in S3 is read-only, it cannot be modified (at least not cheaply). 

I think this schema would make a lot of sense and would not be an humongous effort to implement. Yamcs already supports partitioning per day (or month or year), so easiest would be to move an entire partition to object storage.

haisamido, 2024-07-15:
I think a versatile approach would be to not lock it into s3, but to have a URI construct, that allows one to provide the location of the data for the relevant protocol, e.g.:

```bash
s3://
file:///
sftp://
etc.
```
`file:/// ` could be very useful since this could be an NFS mounted path or even a mounted `s3` 'filesystem'

haisamido, 2024-07-15:
A data partition that we have found very useful is:

```bash
protocol://<root path[bucket] for mission>/data/${yyyy}/${ddd}/${apid}/
 └──  ${mission}_${apid}_${yyyy0}-${ddd0}-${hh0}:${mm0}:${ss0}_${yyyyf}-${dddf}-${hhf}:${mmf}:${ssf}.packets.lz4
```

xpromache, 2024-07-15:
Ok but if we ever implement that, it would contain also the name of the table "tm" and the data will contain additional stuff like generation time, reception time, earth reception time, and some other columns that are in the Yamcs tm table.

We also have in Ymacs a "pp" table that stores directly parameters, that would also benefit of this feature, as other tables like command history, alarms etc.

haisamido, 2024-07-16:
Where is the reference dates and reference times scales stored per APID? It's not unusual for the same spacecraft to have differing reference dates and reference times. Additionally the reference dates will vary as per changes in the mission database and for rollover of counters.

xpromache, 2024-07-17:
> Where are the reference dates and reference times scales stored per APID? It's not unusual for the same spacecraft to have differing reference dates and reference times. Additionally the reference dates will vary as per changes in the mission database and for when there is a rollover of the counters for bit limited parameters

We expect the tm table to contain a UTC correlated time (in fact Yamcs internal time which is referenced to TAI rather than UTC). How the pre-processor generates that time is mission specific. But we do have a time correlation service that helps with that:
https://docs.yamcs.org/yamcs-server-manual/services/instance/time-correlation/

The tm table contains a 64bit integer column "obt" which can be used to store the on-board time for each packet (remember how to store time inside even CCSDS packets is not specified by CCSDS) and the time correlation service maintains its own table with correspondence between obt and Yamcs time.

haisamido, 2024-07-17:
> We expect the tm table to contain a UTC correlated time (in fact Yamcs internal time which is referenced to TAI rather than UTC). How the pre-processor generates that time is mission specific. But [...]

Thanks. Where is the reference date and time stored for each APID? Designating it as UTC or TAI is not always sufficient since the mission may have decided to use a different reference date and time. E.g. TAI's "standard" reference date and time is 1958-01-01T00:00:00; however, several missions I have worked with the, the reference data and time was set to something like 2000-01-01T00:00:00 to reduce the storage requirement.

xpromache, 2024-07-18:
> Thanks. Where is the reference date and time stored for each APID? Designating it as UTC or TAI is not always sufficient since the mission may have decided to use a different reference date and time. E.g. TAI's "standard" reference date and time is 1958-01-01T00:00:00; however, several missions I have worked with the reference date and time was set to something like 2000-01-01T00:00:00 to reduce the storage requirement onboard the spacecraft, presumably.

What I am saying is that the data is supposed to be converted to Yamcs time representation when injected in the tm table. 
Yamcs pre-processors can use a timedecoder which understands some reference epochs but that is used only when injecting the data in Yamcs:
https://docs.yamcs.org/yamcs-server-manual/links/packet-preprocessor/#packet-pre-processing

 Once injected, Yamcs will use the gentime column in the tm table to sort and retrieve the data.