yamcs/yamcs #905: Can yamcs read SFDU data files?

haisamido, 2024-06-25:
Can yamcs read CCSDS SFDU [STANDARD FORMATTED DATA UNITS — STRUCTURE AND CONSTRUCTION RULES](https://public.ccsds.org/Pubs/620x0b2.pdf) files and extract telemetry from the packets?

https://github.com/shanecray

xpromache, 2024-06-26:
No, it cannot.

I am not familiar with SFDU - do the SFDU files embed telemetry packets? Or directly parameters? Or both?

I looked a bit on the internet for SFDU and I can find mostly old references; not even a python library to read the files which is rather odd. May I ask in what context do you need that?

haisamido, 2024-06-26:
At the center I work in most of the telemetry and commanding streams and files are in SFDU.  The SFDU's contain CCSDS Space Packets, after all the frames and encapsulations are taken care of. We are contemplating the evaluation of yamcs and need to know the current capabilities.

xpromache, 2024-06-27:
Ok, so Yamcs cannot read currently SFDU but I guess someone can implement a reader without too much difficulty if it's just to read the binary CCSDS space packets.

However, do those SFDU files contain also the description of the data - like every parameter inside the CCSDS space packets (what we call in Yamcs the Mission Database)? If that's the case, then it will involve quite a bit of work to import those in Yamcs such that Yamcs can decode the packets into parameters.

haisamido, 2024-07-11:
SFDU's which contain CCSDS space packets will not have descriptions of the CCSDS space packets. That would need to be retrieved from the Mission Database.