yamcs/yamcs #906: Array arguments missing from commands in yamcs-web

AdamElliott-AB, 2024-06-26:
Array arguments of a command are missing from the "Send a command" page. Browser console shows error:
  "ERROR TypeError: Cannot read properties of undefined (reading 'ɵcmp')"

How to reproduce:
1. clone yamcs/quickstart
2. edit src/main/yamcs/mdb/xtce.xml to have a command with an Array type argument
3. run yamcs and try to send the above command

[xtce.txt](https://github.com/user-attachments/files/15995216/xtce.txt)

AdamElliott-AB, 2024-06-26:
Bug seems to have been introduced in version 5.9.4
5.9.3 does not exhibit the problem

AdamElliott-AB, 2024-06-26:
This was tested in chrome Version 126.0.6478.114 on Ubuntu

fqqb, 2024-06-27:
Thanks, addressed in pending PR #906. The issue came from a circular import.

AdamElliott-AB, 2024-06-27:
Thank you @fqqb

closed 2024-06-27 by AdamElliott-AB