yamcs/yamcs #909: Support for BlockMetaCommand

tokiedokie, 2024-07-01:
Does yamcs support BlockMetaCommand of XTCE?

I have xtce xml like below. However, I cannot see this BlockMetaCommand from the web commanding page(http://localhost:8090/commanding).
```
        <xtce:BlockMetaCommand name="TimedValveCmdFixed">
                <xtce:MetaCommandStepList>
                    <xtce:MetaCommandStep metaCommandRef="ValveCmd">
                        <xtce:ArgumentAssigmentList>
                            <xtce:ArgumentAssignment argumentValue="ON" argumentName="ValveState" />
                        </xtce:ArgumentAssigmentList>
                    </xtce:MetaCommandStep>
                    <xtce:MetaCommandStep metaCommandRef="NOOP" />
                    <xtce:MetaCommandStep metaCommandRef="NOOP" />
                    <xtce:MetaCommandStep metaCommandRef="NOOP" />
                    <xtce:MetaCommandStep metaCommandRef="NOOP" />
                    <xtce:MetaCommandStep metaCommandRef="NOOP" />
                    <xtce:MetaCommandStep metaCommandRef="ValveCmd">
                        <xtce:ArgumentAssigmentList>
                            <xtce:ArgumentAssignment argumentValue="OFF" argumentName="ValveState" />
                        </xtce:ArgumentAssigmentList>
                    </xtce:MetaCommandStep>
                </xtce:MetaCommandStepList>
            </xtce:BlockMetaCommand>
```

xpromache, 2024-07-01:
Sorry, not supported. 
You can use command stacks for now.

tokiedokie, 2024-07-02:
Thank you for the reply.
Are there any plans to support BlockMetaCommand as command stacks?

xpromache, 2024-07-02:
I think not any time soon unless someone is willing to pay for such feature.