yamcs/yamcs #910: Historical Data not displayed in the WebUI

PalfalviJozsefRemred, 2024-07-02:
Running the quickstart I encountered the following strange behavior.
OS: Windows 10

1. Start yamcs quickstart "myprojct" instance.
2. Start generating simulated data with simulator.py.
3. Observe Telemetry\Paramezers: parameter Battery1_Temp: Everything works as expected, the "Historical Data" and "Charts" display the expected data.
4. Shut down yamcs by closing the server insrance
5. Restart yamcs and open the myproject instance.
6. The same parameter as in Step 3. does not displays any data

[image]

But when exporting to CSV, the data is there.

I would expect that the "Historical Data" and "Charts" display the previoulsy recorded values.

PalfalviJozsefRemred, 2024-07-04:
Update:
I've been able to indentify the root cause of the problem, but I still don't know if this is an expected beahavior. For me it seems to be an issue.

If the YAMCS server is shut down "ungracefully" (i.e. closing the server by just closing the command line window, or killing the process, or any crash happens) then the past parameter values are not visible when reopening the server and web client.

I tested 3 scenarios:
- generate data for 2 minutes --> kill server --> data missing\not visible
- generate data --> stop data generation --> wait 2 minutes --> --> kill server --> data missing\not visible
- generate data --> stop server with proper shut-down --> data visible (OK)

So the issue boils down to questions:
1. Why aren't the past parameter values visible in case the server was killed?
E.g. what happens if the user collects data for several minutes and then acidentally closes the server? Is the data corrupted\lost?

2. And why are the parameter values still there when exporting to CSV?

3. When do the parameters get persisted properly, is there e.g. a timeout?

xpromache, 2024-07-04:
The data shown in plots is coming from the Parameter Archive. When you export CSV it is doing a replay - it extracts packets from the stream archive and reprocesses them - I think we should make this configurable but anyway this is how it is now.

The Parameter Archive is build in segments - whereas the stream archive store a packet consisting of a bunch of parameters (t1, p1,p2,...) (t2, p1,p2,...) at one given time as one record, the parameter archive store multiple values for one parameter ((t1, p1) ,(t2, p1) ,(t3, p1) ,(t4, p1)..), ((t1, p2) ,(t2, p2) ,(t3, p2) ,(t4, p2)...) as different records.
(in db lingo the Parameter Archive is a column oritented database whereas the stream archive is a row oriented database)

The reason is that
1. data for one parameter can be compressed very well (especially if the value doesn't change much which is often the case) and
2.   when retrieving data for a long interval it can be done much faster (i.e. plot a temperature over one year).

However when building this Parameter Archive, Yamcs requires to store in memory an entire segment before writing it to the disk. In the Stream Archive a packet can be flushed to the database as soon as it is received.

There are two ways to build the Parameter Archive: realtime and backfilling.

In EGSE kind of environments we usually do realtime because data cannot come in the past.

In operations we do backfilling because data can often be received in the past (for example a LEO satellite may dump during a 10min pass the data from the last orbits in parallel with realtime data).

The quickstart example uses the realtime filler that builds the Parameter Archive in memory and from time to time it flushes it to the archive. If you kill it, it does not manage to flush it and that's the thing you observe.

To fix it you can run the command yamcs parameter-archive rebuild. That will rebuild the parameter archive from the packets (practically same thing like the CSV export):
https://docs.yamcs.org/yamcs-cli/yamcs_parameter-archive/

Maybe/probably we should detect at startup if the database has not been shutdown cleanly and rebuild it automatically but this is not implemented yet.

Note that I'm working now on moving the RocksDB database  out of Yamcs as a separate process (it will be an optional feature), so I expect this kind of things to happen less often (people are used to long running database processes treated nicely without abrupt kills).

xpromache, 2024-07-04:
I did not answer this question:
> When do the parameters get persisted properly, is there e.g. a timeout?

Currently the automatic flushing is disabled (so segments are flushed only when full or when an interval ends or if the server is shutdown cleanly) and the reason is some performance problems when flushing too often (practically RocksDB does not like to remove data and we have to remove old segments before overwriting with new ones). In the quest to fix this problem I decided to restructure a bit the way the archive is build by using the so called merge operator:
https://github.com/facebook/rocksdb/wiki/merge-operator

I will release that in 5.10 hopefully in some weeks and I will add back the flush interval configuration for the realtime filler.

closed 2025-03-04 by xpromache