yamcs/yamcs #913: ROCKDSDB SST file is ahead of WALs

lorenzo-gomez-windhover, 2024-07-04:
Hello,

While restarting YAMCS, I found the following issue:

```
11:09:00.049 _global [1] RdbStorageEngine Cannot load tablespace fsw: java.io.IOException: org.rocksdb.RocksDBException: SST file is ahead of WALs in CF default
	org.yamcs.yarch.rocksdb.RDBFactory.rdb(RDBFactory.java:112)
	org.yamcs.yarch.rocksdb.RDBFactory.getRdb(RDBFactory.java:64)
	org.yamcs.yarch.rocksdb.Tablespace.loadDb(Tablespace.java:142)
	org.yamcs.yarch.rocksdb.RdbStorageEngine.createTablespace(RdbStorageEngine.java:168)
	org.yamcs.yarch.rocksdb.RdbStorageEngine.getTablespace(RdbStorageEngine.java:141)
	org.yamcs.yarch.rocksdb.RdbStorageEngine.loadTables(RdbStorageEngine.java:270)
	org.yamcs.yarch.YarchDatabaseInstance.loadTables(YarchDatabaseInstance.java:188)
	org.yamcs.yarch.YarchDatabaseInstance.<init>(YarchDatabaseInstance.java:94)
	org.yamcs.yarch.YarchDatabase.getInstance(YarchDatabase.java:97)
	org.yamcs.StreamInitializer.<init>(StreamInitializer.java:49)
	org.yamcs.StreamInitializer.createStreams(StreamInitializer.java:40)
	org.yamcs.YamcsServerInstance.doInit(YamcsServerInstance.java:138)
	org.yamcs.YamcsInstanceService.initAsync(YamcsInstanceService.java:61)
	org.yamcs.YamcsServerInstance.init(YamcsServerInstance.java:122)
	org.yamcs.YamcsServer.addInstance(YamcsServer.java:570)
	org.yamcs.YamcsServer.addGlobalServicesAndInstances(YamcsServer.java:1353)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1177)
	org.yamcs.YamcsServer.main(YamcsServer.java:973)
Caused by: org.rocksdb.RocksDBException: SST file is ahead of WALs in CF default
	org.rocksdb.RocksDB.open(Native Method)
	org.rocksdb.RocksDB.open(RocksDB.java:307)
	org.yamcs.yarch.rocksdb.YRDB.<init>(YRDB.java:146)
	org.yamcs.yarch.rocksdb.RDBFactory.rdb(RDBFactory.java:109)
	org.yamcs.yarch.rocksdb.RDBFactory.getRdb(RDBFactory.java:64)
	org.yamcs.yarch.rocksdb.Tablespace.loadDb(Tablespace.java:142)
	org.yamcs.yarch.rocksdb.RdbStorageEngine.createTablespace(RdbStorageEngine.java:168)
	org.yamcs.yarch.rocksdb.RdbStorageEngine.getTablespace(RdbStorageEngine.java:141)
	org.yamcs.yarch.rocksdb.RdbStorageEngine.loadTables(RdbStorageEngine.java:270)
	org.yamcs.yarch.YarchDatabaseInstance.loadTables(YarchDatabaseInstance.java:188)
	org.yamcs.yarch.YarchDatabaseInstance.<init>(YarchDatabaseInstance.java:94)
	org.yamcs.yarch.YarchDatabase.getInstance(YarchDatabase.java:97)
	org.yamcs.StreamInitializer.<init>(StreamInitializer.java:49)
	org.yamcs.StreamInitializer.createStreams(StreamInitializer.java:40)
	org.yamcs.YamcsServerInstance.doInit(YamcsServerInstance.java:138)
	org.yamcs.YamcsInstanceService.initAsync(YamcsInstanceService.java:61)
	org.yamcs.YamcsServerInstance.init(YamcsServerInstance.java:122)
	org.yamcs.YamcsServer.addInstance(YamcsServer.java:570)
	org.yamcs.YamcsServer.addGlobalServicesAndInstances(YamcsServer.java:1353)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1177)

```

I looked into this and found several issues on the internet and no one (including me) seems to understand the root cause of this, which worries me a bit....

I did find a comment that pointed me in somewhat the right direction:https://github.com/arangodb/arangodb/issues/9757#issuecomment-631506502

I started looking at the YAMCS RocksDB interfacing code and did find the options to ignore the corrupted records(though I don't think it is exposed at the moment to YAMCS admins/users). I modified yamcs-core(you can see the changes [here](https://github.com/WindhoverLabs/yamcs/commit/959270c3a748d92d1a3d159001798b0602a0b53b#diff-fb18b0e9863d910a111e9bb2beefe01a06240e7a0532fcaaee3f7cf18ed54482R108)) and set it to ignore the corrupted WAL records with `dbOptions.setWalRecoveryMode(WALRecoveryMode.SkipAnyCorruptedRecords);`. 

I restarted YAMCS and this "fixed" the issue, though I really don't like this fix...

I really want to understand the root cause of this....Is it one of my plugins? A data race in the RocksDB API?

Do you guys have any insight into what might be the root cause of this?

Also, how open are you to exposing the "WALRecoveryMode" option to the RocksDB YAML for yamcs users to be able to at the very least have an alternative to "recover" from this error and ensure they can have a chance at salvaging their yamcs archive.

Hope this is clear enough
Lorenzo

xpromache, 2024-07-05:
> I did find a comment that pointed me in somewhat the right direction:arangodb/arangodb#9757

Hi Lorenzo,

I have never seen this error before. I see that running out of space on
disk may trigger it. Could that have been the case for you?

As for fixing it, can't the ldb tool offered by Rocksdb fix it? We can add
an option in Yamcs but only if there is no other option.

nicolae

xpromache, 2024-10-08:
closing this as no news