yamcs/yamcs #922: yamcs-web feature request: hyperlinks for nested containers in mdb

0lionelzhang0, 2024-08-02:
Hello,

Some members from our team were wondering if it would be possible to implement hyperlinks for nested container definitions. 

Currently in 5.9.8, when you open a container definition in MDB the nested containers are listed as plain text and not clickable. This makes it a little cumbersome to traverse through a container's definition.

Our XTCE looks something like this:

``` 
<ContainerSet>
	<SequenceContainer>
		<EntryList>
			<ContainerRefEntry containerRef="ContainerA" />	
```

[image]

Thanks!
Lionel

fqqb, 2024-08-07:
Indeed, shall add this.

fqqb, 2024-08-11:
On master, and in v5.10.0

closed 2024-08-11 by fqqb