yamcs/yamcs #927: Issues with enabling the yamcs-sle plugin

0x4ndy, 2024-08-20:
I've been trying to enable the `yamcs-sle` plugin on my `yamcs` instance. Looking at the doc, it should be as simple as adding it as a dependency to my `pom.xml` file. However, this doesn't seem to work for me at all. I'm not sure how to troubleshoot this, but it looks like Maven doesn't even download the dependency. Can you please give me some hints on how to troubleshoot this issue? Thanks in advance!

xpromache, 2024-08-20:
if you run mvn yamcs:run in this repo:
https://github.com/yamcs/yamcs-sle

does it work?

The simulator of Yamcs does not support SLE. What you can do though is to run the sle to UDP bridge:
https://github.com/yamcs/jsle

so Yamcs connects to the bridge via SLE and the bridge communicates to the simulator with UDP.

0x4ndy, 2024-08-20:
Yes, that makes sense. yamcs-sle on standalone works fine.

closed 2024-08-20 by 0x4ndy