yamcs/yamcs #931: TSE Updates/Commanding from MDB Algorithm

nmaas87, 2024-09-07:
Dear all,
I need to regularly get data from my TSE. For that I prepared an Algorithm in my MDB which fires OnPerodicRate, but I cannot find how to trigger a Yamcs/TSE command from the MDB Algorithm. Could you kindly point me in the right direction?
Thank you :)

fqqb, 2024-09-17:
There's no utility function for that right now. But it's an interesting idea.
For now, probably best to trigger outside of Yamcs. Or within Yamcs from a custom service in Java.

nmaas87, 2024-09-17:
For the moment I wrote an external script which subscribes to a local boolean parameter with which I can control within said script if it shall command via Yamcs the TSE every x timeslots or not. A little bit clunky, but works for the moment. But yes, as SCPI does wait for an input/trigger to deliver values, it would be nice to have something like this within the TSE to allow for e.g. refreshing power supply or multimeter readings regularly.

And a MQTT link for TSE would also be awesome :D