yamcs/yamcs #933: Trying to understand how my TC parameters are encoded.

0x4ndy, 2024-09-09:
So, I have a simple TC with one parameter (unsigned integer). I provide a value `43690` decimal (which is a nice 4 bytes integer, hex: `AAAA` and bin: `1010101010101010`) and send it. 
Once the command is sent, I get the following info:
Unprocessed binary: `180cc00000000000000000220000012caa`
Binary: `180cc013000d5409649eff220000012caa003b21`
I'm having trouble understanding where the value of my parameter is in this payload.

On the receiving end (I use SLE) I get the following:
Binary: `20ab001a00180cc013000d5409649eff220000012caa003b213f67`
Hex: `01000001010101100000000000110100000000000011000000011001100000000010011000000000000110101010100000010010110010010011110111111110010001000000000000000000000000100101100101010100000000000111011001000010011111101100111`
This seems to be what YaMCS sent, but again, I don't see where the value of my parameter is. 

Here are some screenshots.
Preparing a TC:
[image]
TC sent:
[image]

xpromache, 2024-09-09:
You have to paste your command and argument (we call those arguments rather than parameters) definition, otherwise it's impossible to tell.

0x4ndy, 2024-09-09:
Do you mean the TC definition? I use the default SLE one from: https://github.com/yamcs/yamcs-sle/blob/master/src/main/yamcs/mdb/sle.xls

[image]

xpromache, 2024-09-09:
yeah, that looks like a bug. Also note that your 4 bytes 0xAAAA are in fact only 2....

xpromache, 2024-09-09:
Also note that the uint8 type is in fact just the name of the type described in the DataTypes. I think the problem is that the encoding is on 8 bits but the value itself is not restricted to 8 bits.
It is a limitation of the spreadsheet loader where you cannot specify the size of the engineering value (it is possible in the XTCE definition).

xpromache, 2024-09-09:
To elaborate some more: it is possible that you have a 32 bit integer engineering value which gets encoded to a 8 bit raw value by using some calibration of some sort.
It is clearly not the case here (because there is no calibrator defined) but Yamcs doesn't have this kind of validation built in (yet).

0x4ndy, 2024-09-09:
Yes, it's 2 bytes instead of 4, my mistake. 

Oh, so maybe it's more like a feature instead of a bug at this point. I think we can close this issue though.

closed 2024-09-09 by 0x4ndy