yamcs/yamcs #934: Export CSV results in incorrect data if packet has a ContainerRefEntry

AdamElliott-AB, 2024-09-11:
When using ContainerRefEntry, all packet parameters are parsed correctly when telemetry is received, however when exporting to .csv for any parameters after the ContainerRefEntry then data in the .csv is corrupted.

How to reproduce:

1. Pull the latest yamcs quckstart
2. Replace src/main/yamcs/mdb/xtce.xml with the contents of the attached file [xtce.txt](https://github.com/user-attachments/files/16969248/xtce.txt) (same as original except with first 2 parameters of the "Spacecraft" packet moved into a separate SequenceContainer)
3. compile and run yamcs
4. run simulator.py (no modification needed)
5. View Historical Data for the first parameter after the ContainerRefEntry (/myproject/A) - should be correct
6. Export /myproject/A to csv - parameter values in the csv are incorrect

closed 2024-09-21 by xpromache

xpromache, 2024-09-21:
Thanks for the detailed report. Fixed on master.

The reason the csv was showing different output than the Historical Data is that the csv was running a replay with only that parameter as part of extraction. Yamcs tries to be smart and only extract the parameters requested. In this case, because the location of that parameter is relative to the previous container, it should have extracted all the parameters from the container but it wasn't - this is the bug which I fixed. It can work more efficient when it has absolute positions of parameters in containers rather than relative.

Those replays used for extraction (with the associated optimizations) come from the time when we did not have a parameter archive. 
@fqqb we should make all those queries use the parameter archive unless the "source=replay" is specified and then maybe we can have that as a option either in the export dialog or a general user preference.

AdamElliott-AB, 2024-09-23:
Thank you very much for the fix @xpromache

commit a460b512 (2024-09-21): Fixes #934