yamcs/yamcs #935: Booleans in mission database?

0x4ndy, 2024-09-12:
I'm trying to define a boolean type for a command argument in the mission database. I've created an argument with boolean as both engineering and raw value, hoping that the UI will nicely display True/False radio boxes and the argument itself will be encoded to one bit 0/1. 
Here's my definition:
[image]

Unfortunately, when trying to encode this parameter, YaMCS gives me the following error: `DataEncoding BooleanDataEncoding(sizeInBits:1) not implemented`

I checked that if I define the raw value as a string, the parameter is encoded as a True/False string value. Although this works fine, that's not what I want. 
Is there a way to define a boolean command argument with the engineering value True/False that can still be encoded to a single bit 1/0?

xpromache, 2024-09-12:
try using uint encoding on 1 bit.

0x4ndy, 2024-09-12:
I've tried that just now, but then I have to shift my next argument by 7 bits. Otherwise, I get an error that my parameter doesn't start at the byte boundary. So here's command my definition:
[image]
And here's my boolean definition:
[image]

My issue now is that this boolean parameter is encoded as two bytes (?), which is messing up the actual value. Here's what I'm getting:
```
' \xab\x002\x00\x18\x0c\xc0\x02\x00%T\rv\xc6\xda"\x00\x00\x01,NoneRT\x00aaa\x00bbb\x00\x80uniqueId\x00\x00\xf0*Rr'
```
That boolean parameter was encoded to `\x00\x80` (which is `10000000` binary), instead of `\x01`. Even if 2 bytes was was ok, it still should be `\x01\x00`. What am I missing?

xpromache, 2024-09-12:
hmm, I'm surprised you don't get an exception due to the next string not being byte aligned.

I think you should use uint encoding on 8 bits to make the boolean fill-up the whole byte.

0x4ndy, 2024-09-12:
Oh, yes you're right. Shifting the boolean arg worked fine. Indeed the `\x00` byte belongs to the string arg before. All works fine. Thanks for your help!

closed 2024-09-12 by 0x4ndy