yamcs/yamcs #937: SIGILL with latest YAMCS release 5.10.1 (RockDB related)

snthzm, 2024-09-13:
Hello, 
Just pulled the latest YAMCS version to patch an error with Array display on the web interface, but the "./run_example.sh pus" command doesn't work anymore, am I the only one ? Any possible reason in mind why this is happening ? 
```
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0x00007f6712039c5e, pid=26449, tid=26450
#
# JRE version: OpenJDK Runtime Environment (21.0.4+7) (build 21.0.4+7-Ubuntu-1ubuntu222.04)
# Java VM: OpenJDK 64-Bit Server VM (21.0.4+7-Ubuntu-1ubuntu222.04, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# C  [librocksdbjni7115362483400843214.so+0x439c5e]  rocksdb::ShardedCacheBase::ShardedCacheBase(rocksdb::ShardedCacheOptions const&)+0x7e
```

xpromache, 2024-09-13:
I just discovered this yesterday myself :(

What CPU are you running on?

It is from the Rocksdb library which in the latest version is compiled by
us and does not support old CPUs. Yesterday I compiled yesterday a version
that works on CPUs as old as Intel Core2. To use it you have to override
the library pulled in by default by Yamcs. You can use this in the pom.xml:
  <dependencyManagement>
   <dependencies>
       
       <dependency>
           <groupId>org.yamcs</groupId>
           <artifactId>yamcs-rocksdb</artifactId>
           <version>9.4.0.9-core2</version>
       </dependency>
   </dependencies>
  </dependencyManagement>

Maybe we will move the default version to this, still thinking about it...

snthzm, 2024-09-16:
Hi @xpromache,
Forcing the version of the dependency fixes the error yes (My CPU is a Intel i7-3630QM). If the forced version is enough, it makes sense to have it as default, but at least the issue is known.
Thanks

closed 2024-09-16 by snthzm

xpromache, 2024-09-16:
The guys from rocksdb were suggesting to compile with -mhaswell as a good default and I was thinking to do just that (the current scheme is not good because it is whatever the github runners run with...).

I wonder if -mhaswell will work with your CPU which I believe is Ivy Bridge, one generation older:
https://gcc.gnu.org/onlinedocs/gcc/x86-Options.html

I will give it a try and kindly ask you to test one of the following days.

xpromache, 2024-09-22:
For the record: I run some benchmarks with the core2 compatible rocksdb versus the non-compatible version (the one causing SIGKILL on older platforms).  
The benchmark consisted in loading about 5GB of data in the pp table, building the parameter archive and retrieving all parameters from the parameter archive.

I ran the benchmark on several recent CPUs and I couldn't see any performance benefit in using the library supporting the newer architecture vs the core2 version. 

Starting from the next Yamcs version I will make the core2 rocksdb the default.