yamcs/yamcs #951: Manual usage of database

codenamefrance, 2024-10-08:
Hi, 
I'm trying to setup a YAMCS-based tool to achieve telemetry from a car and compare it with historical telemetry of another experiement. I was wondering what are yamcs' potentials in doing this: after collecting some telemetry data, how can I "store" them in order not to compromise them with new ones and, eventually, re-utilise them in another situation?

xpromache, 2024-10-08:
Why do you want to use Yamcs for this task? 

For the moment it is not very easy to import telemetry. Most of the time we require to have before hand the "Mission database" - that is a description of all the data that is going to be in the archive. So not really convenient. We are making that more flexible but we are not there yet.

I guess one of the many timeseries databases like InfluxDB,  TimescaleDB or DuckDB will work better for your scenario. Their query capabilities are also much better than what we have in Yamcs.

codenamefrance, 2024-10-08:
My idea was to have only one system which is capable to handle everything. Is there a way to store all telemetry from an experiment? Even if I can't import that back, at least to retrieve it.

xpromache, 2024-10-08:
Maybe have a look at this method:
https://docs.yamcs.org/python-yamcs-client/general/client/#yamcs.client.YamcsClient.load_parameter_values

You will need a Yamcs MDB configuration with a writable path, something like this in the yamcs.<instance>.yaml
```
...
mdb:
    ....
    - type: "xtce"
      args:
          file: "mdb/writable_subsys.xml"
      writable: true
```

where the file writable_subsys.xml has to exist and be an empty spacesystem like:
```
<?xml version="1.0" encoding="UTF-8"?>
<SpaceSystem xmlns="http://www.omg.org/spec/XTCE/20180204" name="writable_subsys">
</SpaceSystem>
```

Then all parameter names you publish with the python call above have to start with "/writable_subsys/" (which of course you can change to whatever you want).

codenamefrance, 2024-10-08:
I see yamcs uses a rockdb database to store data. Is it possible to access it for other purposes? (e.g. store other kind of data?)

codenamefrance, 2024-10-08:
In your opinion is it possible to exploit yamcs' rockdb to store other datas?

codenamefrance, 2024-10-08:
Is there a way to get around this by some external script? My idea is: I need a db to store few data; as yamcs already use a db, I'd like to use it by querying it on my own (e.g. with python)
Do you have some refs or keywords to search in order to achieve this?

xpromache, 2024-10-08:
No, RocksDB is a (key,value) database, it has no querying capabilities - practically you have a get, a put and an iterator (with some variations).

Everything implemented on top of it is part of Yamcs.

codenamefrance, 2024-10-08:
Where can I see how yamcs handles this?

xpromache, 2024-10-08:
search for rocksdb in the sourcecode?
You can start here:
https://github.com/yamcs/yamcs/tree/master/yamcs-core/src/main/java/org/yamcs/yarch/rocksdb

closed 2025-03-04 by xpromache