yamcs/yamcs #952: Multiple Data Streams in Same Instance

SadLinux, 2024-10-08:
Hello,
 
I am trying to setup multiple data streams in the same instance. For example, let the instance be called "East Area" Within that instance, I have multiple test positions each with their own stream. Within their packet stream there is an identifier. I am able to reference this byte within the xcte.xml file using a base container that has RestrictionCriteria. However, since each test position has different telemetry data I want to setup a .xml file for each. like so:
mdb:
  # for test stand 1
  - type: xtce
    writable: true
    args:
      file: mdb/xtce.xml
 # for test stand 2
  - type: xtce
    writable: true
    args:
      file: mdb/test2.xml
I have found that the restrictioncriteria only works for the xtce.xml. However, every test stand has different telemetry data so I believe I need to use different xml files for each. Am I missing something?

Here is the example of the restriction criteria: 
		 <SequenceContainer name="Test Stand 1">
		 				
						<BaseContainer containerRef="DSU">
					<RestrictionCriteria>
						<ComparisonList>
							<Comparison value="0" parameterRef="Stream" />
						</ComparisonList>
					</RestrictionCriteria>
				</BaseContainer>
		</SequenceContainer>

Thank you for your time.

xpromache, 2024-10-08:
If what you call stream is the same with Yamcs stream, then you have a property rootContainer which tells Yamcs which container it should use to start processing data from that stream:
https://docs.yamcs.org/yamcs-server-manual/data-management/streams/

If what you call stream is something unrelated to Yamcs, then you likely need to setup different data links in Yamcs, one for each of your Data streams, and each link will have its own Yamcs stream which you can then configure with the rootContainer property.

SadLinux, 2024-10-08:
Okay, I don't quite understand how to define the rootcontainer. This is the directory to the xml files C:\myproject\src\main\yamcs\mdb
within the mdb folder lives xtce.xml and test2.xml. Am I defining the root container as the sequence container in test2.xml or is it something else?

Sorry, I'm newer to yamcs.

Thank you for your time, xpromache.

xpromache, 2024-10-08:
I guess you have two containers in xtce.xml and the test2.xml which have no parent (no BaseContainer) - those are your root containers. 

You have to define them in the stream configuration, for example:
https://github.com/yamcs/yamcs/blob/master/examples/simulation/src/main/yamcs/etc/yamcs.simulator.yaml#L131

The stream tm_realtime doesn't define a rootContainer, in that case Yamcs takes the first one it finds in the MDB (so that's not really a safe configuration, better define one explicitly).

Then the next problem is to arrange the two telemetry streams to come on the two Yamcs streams. In the example above there are two TCP data links doing that.

SadLinux, 2024-10-16:
Thank you for your help I was able to get it figured out.

closed 2024-10-16 by SadLinux