yamcs/yamcs #956: CFDP incoming transfer becomes "stuck" after receiving EOF while "paused/suspended". NAK doesn't get sent to request missing data

0lionelzhang0, 2024-10-10:
Yamcs version: 5.10.3

Scenario:
- Downlink transaction from remote to Yamcs
- Remote and Yamcs lose connection before remaining FileData and EOF received (e.g. end of ground station pass)
- Remote periodically resends EOF 
- Yamcs transaction goes into paused state after inactivity timeout.
- Upon next connection, Yamcs receives EOF while in the suspended state
  - EOF ACK is sent to remote
  - eofPacket becomes not null
  - NAK is NOT sent or scheduled yet because still suspended
- User resumes the incoming transfer in Yamcs
- Remote sends another EOF

Current behavior:
- EOF packet is [ignored by Yamcs](https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/main/java/org/yamcs/cfdp/CfdpIncomingTransfer.java#L346) because eofPacket is not null
- NAK is never sent or scheduled by sendOrScheduleNak because the initial call was done in "suspended" state

Expected behavior:
- Yamcs sends NAK to remote to request missing data

Possible fix:
- The logic in [sendOrScheduleNak](https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/main/java/org/yamcs/cfdp/CfdpIncomingTransfer.java#L172-L191) can be changed to something like this instead so that NAKs are still scheduled if suspended, but not sent until resumed.
```
    private void sendOrScheduleNak() {
        if (inTxState != InTxState.RECEIVING_DATA) {
            return;
        }

        if (!eofReceived() && !immediateNak) {
            return;
        }

        long now = System.currentTimeMillis();
        if (now - lastNakSentTime < nakTimeout) {
            return;
        }

        if (!suspended && sendNak()) {
            lastNakSentTime = now;
        }

        executor.schedule(this::sendOrScheduleNak, nakTimeout, TimeUnit.MILLISECONDS);
    }
```

This issue does not occur if Yamcs transaction never enters suspended state, so the current workaround we use is to increase inactivityTimeout.

Thanks,
Lionel

xpromache, 2024-10-10:
Can you please check the cfdp-fixes branch to see if it fixes this issue?
https://github.com/yamcs/yamcs/tree/cfdp-fixes

I have recently noticed that Yamcs sends ACKs and FIN while paused and this is not ok.

0lionelzhang0, 2024-10-10:
Thank you for the quick response! Just tested with your branch and it does seem to fix it

xpromache, 2024-10-11:
In my branch, the receiver ignores the EOF received while suspended, thinking that since the EOF ACK cannot be send while Suspended,  the sender will have to send the EOF again after resuming.

 However, thinking more about it I now believe it is not the correct behavior. Reading the [standard](https://public.ccsds.org/Pubs/727x0b5e1.pdf) again on page 4-24:

> 
> On Notice of Suspension of the Copy File procedure, the receiving CFDP entity shall
> a) suspend transmission of NAK PDUs;
> b) suspend any transmission of Keep Alive PDUs;
> c) suspend the inactivity timer;
> d) suspend transmission of Finished (complete) PDUs;
> e) suspend the application of Positive Acknowledgment Procedures to PDUs previously issued by this entity;
> f) issue a Suspended.indication if so configured in the MIB (see table 8-1);
> g) save the status of the transaction.
> 

It does not say that EOF should not be processed or that  EOF ACKs should not be send.

And it makes sense: if the EOF is received while suspended and that completes the file, upon resuming, the Receiver can immediately send a FIN packet which will indicate to the Sender that the file has been received correctly, even though the Sender hasn't received the EOF ACK.

So I will change the code to:
-  not ignore the EOF while suspended
-  introduce a new configuration ackEofWhileSuspended  (defaulting to true) which controls the sending of the EOF ACK while suspended
- trigger immediately the NAK or the FIN upon resuming if the EOF has been received while suspended (this should solve your bug).
- the EOF ACK will **not** be sent upon resuming, regardless of the ackEofWhileSuspended setting. 

Comments, suggestions?

xpromache, 2024-10-11:
The use case for that would be if the operator wants that nothing goes out of the CFDP service when suspended (maybe some critical activity takes place and wants to make sure only operator's commands go out).  

In addition, when the EOF is received, if the file is complete then after resuming Yamcs will send directly the FIN which supersedes the EOF ACK. If the file is not complete, it will send a NAK or two so the transfer will continue for a little while, receiving and processing again the EOF will not be a big overhead.

And the final argument is that is very likely the EOF ACK will not be received by the Sender because often the reason for suspension is that the link is down.

0lionelzhang0, 2024-10-11:
Makes sense, I'm good with all these proposed changes. Thank you!

> - not ignore the EOF while suspended
> - introduce a new configuration ackEofWhileSuspended (defaulting to true) which controls the sending of the EOF ACK while suspended
> - trigger immediately the NAK [...]

xpromache, 2024-10-11:
Please check again the cfdp-fixes branch, I implemented the changes there.

I will do some more tests myself but I need to implement the suspend/resume in the simulator for that.

0lionelzhang0, 2024-10-12:
Works great! I'm no longer seeing the stuck behavior.

xpromache, 2024-11-03:
I merged the cfdp-fixes branch that should fix the issue reported.
I still want to change the simulator to allow better testing the suspend/resume and fix a few other CFDP issues (proxy transfer doesn't work as per the standard).
We keep this issue open until the simulator is fixed.

0lionelzhang0, 2024-11-03:
Awesome, thank you, Nicolae!