yamcs/yamcs #957: Sorting "Telemetry -> Packets" by reception time

0lionelzhang0, 2024-10-10:
Hello,

We are wondering if it would be possible to add a toggle option to sort packets by reception time instead of generation time?

Primary use cases are when the generation time is in the future or the past.

Thanks!
Lionel

xpromache, 2024-10-11:
Unfortunately not possible easily. The database is sorted on the generation time and there is no secondary index, so the only way to sort by something else is to read all the data in memory and sort it (or make a temporary table like Mysql does). I don't think we will implement sorting in memory or via temporary table any time soon. 
Maybe it is possible to add a secondary index (we have implemented these a while ago), but I don't remember if we can use secondary indices for sorting.

For this particular case of packet with the wrong time, my idea was that the pre-processor should handle it: if it detects that a packet has obviously a wrong time, it will use the local (Yamcs server) time and set a flag so it can be found later. There is a TmPacket.setLocalGenTimeFlag method for that.

Which pre-processor are you using?

0lionelzhang0, 2024-10-11:
Got it, I will look into the setLocalGenTimeFlag method. How does one pull that flag value at a later time? I was looking through the HTTP and Python API but could find it in the Packet model
https://docs.yamcs.org/python-yamcs-client/tmtc/model/#yamcs.tmtc.model.Packet
https://docs.yamcs.org/yamcs-http-api/packets/list-packets/

We use a custom pre-processor extending AbstractPacketPreprocessor.