yamcs/yamcs #964: Download button in 'Export parameter data' requester no longer working?

kabunkie, 2024-10-15:
While upgrading Yamcs to the most recent version I noticed that starting with V5.9.10 the 'Download' button in the CSV Export requester no longer seems to react to any user interaction. (Tested in Edge and Firefox) 

In V5.9.8 the HTML code looked like this and was/is still working:
`<ya-download-button><button class="ya-button primary"> DOWNLOAD </button><a download="" style="display: none;" href="/api/archive/ziggy:exportParameterValues?parameters=/Ziggy/pm03cnt&amp;delimiter=TAB&amp;start=2024-10-15T08:16:22.166Z&amp;stop=2024-10-15T09:16:22.166Z"></a></ya-download-button>`

In V5.9.10 it changed to this (still the same in V5.10.3) and the button is ignoring any clicks:
`<ya-download-button appearance="primary"><ya-button _nghost-ng-c1143172173=""><button _ngcontent-ng-c1143172173="" class="primary"> DOWNLOAD </button></ya-button><a download="" style="display: none;" href="/api/archive/ziggy:exportParameterValues?parameters=/Ziggy/pm05cnt&amp;delimiter=TAB&amp;start=2024-10-15T08:27:01.253Z&amp;stop=2024-10-15T09:27:01.253Z"></a></ya-download-button>`

The href is still valid and works when selected manually.

HTH

fqqb, 2024-10-15:
Thanks, could reproduce. Fix will be included in upcoming release (both 5.9.x and 5.10.x)

fqqb, 2024-10-16:
On master, and released.

closed 2024-10-16 by fqqb