yamcs/yamcs #967: How to convert packet generation time (TAI) from 52 bit (big endian) to a float?

haisamido, 2024-10-17:
We have the case where the binary container for packets' generation timestamps are defined as 32 bits to the left of the decimal, and 20 bits to the right of decimal.

_**What's the best way for yamcs to convert this to TAI or UTC?**_

@shanecray

xpromache, 2024-10-18:
That's the typical task for a pre-processor. You can use the TimeEncoding class of Yamcs to convert between whatever you read from the packet and the Yamcs time (which is currently UTC+37 leap seconds) . Note that the timestamp associated to packets in Yamcs are only millisecond resolution.

haisamido, 2024-10-23:
> ... Note that the timestamp associated to packets in Yamcs are only millisecond resolution.

Thank you for your reply @xpromache I'm a bit confused with the above statement. We will have our own timestamp resolutions so why would yamcs resolution be an issue?

xpromache, 2024-10-23:
Maybe I meant precision instead of resolution. Basically yamcs stores the packet generation time as milliseconds since 1970 (including leap seconds). If you had some microseconds or nanoseconds in there, they are lost.

We do have a timestamp type that includes picoseconds but is used only in a few places.

haisamido, 2024-10-23:
> .. Basically yamcs stores the packet generation time as milliseconds since 1970 (including leap seconds). If you had some microseconds or nanoseconds in there, they are lost.

That's a bit odd since the our packet's timestamp's sub-seconds are defined in our system so not sure why there is a limitation to precision. Shouldn't the user of yamcs drive the precision?