yamcs/yamcs #972: # A fatal error has been detected by the Java Runtime Environment: #

JohnAdam-123, 2024-10-30:
I have encountered this JRE error when starting the server. Can I get help of how to solve this error.
[image]
  `14:46:26.250 _global [1] YamcsServer Loading service HttpServer
14:46:27.061 _global [1] RdbStorageEngine Creating or loading tablespace _global
14:46:27.070 _global [1] Tablespace [_global] Creating database at yamcs-data/_global.rdb
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0x00007e6a91639c5e, pid=10125, tid=10126
#
# JRE version: OpenJDK Runtime Environment (17.0.12+7) (build 17.0.12+7-Ubuntu-1ubuntu222.04)
# Java VM: OpenJDK 64-Bit Server VM (17.0.12+7-Ubuntu-1ubuntu222.04, mixed mode, sharing, tiered, compressed oops, compressed class ptrs, g1 gc, linux-amd64)
# Problematic frame:
# C  [librocksdbjni6111293489742449277.so+0x439c5e]  rocksdb::ShardedCacheBase::ShardedCacheBase(rocksdb::ShardedCacheOptions const&)+0x7e
#
# Core dump will be written. Default location: Core dumps may be processed with "/usr/share/apport/apport -p%p -s%s -c%c -d%d -P%P -u%u -g%g -- %E" (or dumping to /home/john/Documents/DEMO/projects/openmct-yamcs-master/myproject/target/yamcs/core.10125)
#
# An error report file with more information is saved as:
# /home/john/Documents/DEMO/projects/openmct-yamcs-master/myproject/target/yamcs/hs_err_pid10125.log
#
# If you would like to submit a bug report, please visit:
#   https://bugs.launchpad.net/ubuntu/+source/openjdk-17
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  8.804 s
[INFO] Finished at: 2024-10-30T14:46:27+03:00
`

JohnAdam-123, 2024-10-30:
> This appears when using a 5.10.0-4 version on older computers but should be solved by Yamcs 5.10.5, please give it a try.

So where do I go and change the version, or do I update this?

xpromache, 2024-10-30:
inside the pom.xml which I guess you have in the myproject directory.

JohnAdam-123, 2024-10-30:
> inside the pom.xml which I guess you have in the myproject directory.

Yes, I have it. Let me change and run it again. I'll give you the feedback.

JohnAdam-123, 2024-10-30:
> inside the pom.xml which I guess you have in the myproject directory.

Yes. The update of the version solved the problem from `<yamcsVersion>5.10.1</yamcsVersion>` to `<yamcsVersion>5.10.5</yamcsVersion>`

Thank you.

closed 2024-10-30 by xpromache