yamcs/yamcs #974: [ERROR] Failed to execute goal org.apache.maven.plugins:maven-compiler-plugin:3.13.0:compile (default-compile) on project myproject: Fatal error compiling: error: release version 21 not supported -> [Help 1]

JohnAdam-123, 2024-10-31:
I was trying to compile the YAMCS file server so that I might be able to run the project into my computer, but I ran into this error.
Here is the command I used `. /mvnw compile ` and below is the screenshot output.
[image]

xpromache, 2024-10-31:
make sure you install the jdk package:
apt install openjdk-21-jdk

fqqb, 2024-10-31:
Also see output of `./mvnw -version`

JohnAdam-123, 2024-11-01:
> make sure you install the jdk package: apt install openjdk-21-jdk

I managed to solve it. I had to change the jdk to 17, but also settings the environment path. Then I compiled it again.

JohnAdam-123, 2024-11-01:
> Also see output of `./mvnw -version`

Okay.

closed 2024-11-01 by xpromache