yamcs/yamcs #979: Web UI enum bug when nested in array -> aggregate -> enum

0lionelzhang0, 2024-11-07:
Hello,

Yamcs version 5.10.5

We noticed a minor visual bug when viewing an enumeration parameter nested in an array -> aggregate -> enum. It only shows the number value instead of the string value.

[image]

[image]

Cases that do work as expected (enum shows string value):
- array -> enum
- aggregate -> enum

Thanks,
Lionel

0lionelzhang0, 2024-11-07:
Ahhh yes it is, sorry about that!

closed 2024-11-07 by 0lionelzhang0

xpromache, 2024-11-07:
no problem. 
It's funny we discovered the same bug simultaneously - this bug has been there forever.