yamcs/yamcs #997: Only one subscription gets parameter expiration updates

merose, 2025-01-15:
Found in Yamcs 5.9.8. Will verify with current Yamcs when I get time.

Only one subscription to a parameter will get expiry notifications on that parameter. Other subscriptions will receive parameter updates but not expiry notifications.

To reproduce:
1. Set up a parameter in a container with a defined <DefaultRateInStream> so that the parameter will receive expiration notifications.
2. Set up a data link to feed the container, stopping periodically so that its parameters will expire.
3. Open a web browser to Telemetry > Parameters, and browse to the parameter from step 1. Verify that it expires by identifying the small clock icon that appears when the parameter expires.
4. Open a second web browser to Telemetry Parameters, and browse to the parameter from step 1.

Expected result: Both web browsers show parameter expiration with the small clock icon next to the parameter value.

Actual result: Only one browser window shows expiration.

[image]

shefalijoshi, 2025-01-15:
Also reproduced with yamcs version 5.10.4.

xpromache, 2025-02-03:
Sorry for this problem. I wanted to fix it in the upcoming 5.11.0 release but there are too many changes and I ran out of time. I twill be done in 5.12. sometimes in the next couple of weeks.

The main issue annoying me is the AcquisitionStatus is currently an enum causing a parameter being expired to overwrite the fact that it was invalid.  I will change this status to a bitfield such that it is possible to have a parameter both invalid and expired at the same time.

The bug reported comes from the fact that the status gets duplicated (in order to not lose the validity) and the duplication goes awry at the second copy.

Once I change the status to allow expiration without overwriting the validity,  I will do the expiration at processor level rather than currently done as part of the subscription.

commit bb7b1ac5 (2025-02-03): refactor parameter acquisition status

Instead of having an enum, use a bitfield to allow to have a parameter both expired and invalid at the same time
Fixed #997
closed 2025-02-04 by xpromache