yamcs/yamcs #999: DatabaseCorruptionException for small subset of data

0lionelzhang0, 2025-01-27:
Hello,

We are experiencing the error below (parameter name removed) for a subset of recently ingested data into Yamcs. None of the data from this container can be queried at the moment (shows the error about "Cannot find a time segment for parameterGroupId" on the web app).

Do you know how we can fix this?

Thanks,
Lionel

```
10:56:53.402 _global [95] RouteHandler 65: Responding '500 Internal Server Error': org.yamcs.utils.DatabaseCorruptionException: Cannot find a time segment for parameterGroupId=7 segmentStart = 1737842642437 despite having a value segment for parameterId: Entry [pid=668, fqn=<our-parameter-name>, engType: UINT64, rawType: UINT64]: org.yamcs.http.InternalServerErrorException: org.yamcs.utils.DatabaseCorruptionException: Cannot find a time segment for parameterGroupId=7 segmentStart = 1737842642437 despite having a value segment for parameterId: Entry [pid=668, fqn=<our-parameter-name>, engType: UINT64, rawType: UINT64]
        org.yamcs.http.RouteHandler.handleException(RouteHandler.java:171)
        org.yamcs.http.RouteHandler.dispatch(RouteHandler.java:130)
        org.yamcs.http.RouteHandler.handle(RouteHandler.java:94)
        org.yamcs.http.RouteHandler.channelRead0(RouteHandler.java:66)
        org.yamcs.http.RouteHandler.channelRead0(RouteHandler.java:41)
        io.netty.channel.SimpleChannelInboundHandler.channelRead(SimpleChannelInboundHandler.java:99)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:444)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
        io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:103)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:444)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
        io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:103)
        io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
        org.yamcs.http.HttpRequestHandler.channelRead(HttpRequestHandler.java:125)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:444)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
        io.netty.handler.codec.http.cors.CorsHandler.channelRead(CorsHandler.java:95)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
        io.netty.channel.CombinedChannelDuplexHandler$DelegatingChannelHandlerContext.fireChannelRead(CombinedChannelDuplexHandler.java:436)
        io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:346)
        io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:318)
        io.netty.channel.CombinedChannelDuplexHandler.channelRead(CombinedChannelDuplexHandler.java:251)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
        io.netty.handler.traffic.AbstractTrafficShapingHandler.channelRead(AbstractTrafficShapingHandler.java:506)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
        io.netty.handler.traffic.AbstractTrafficShapingHandler.channelRead(AbstractTrafficShapingHandler.java:506)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:442)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:412)
        io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:440)
        io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:420)
        io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
        io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
        io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:788)
        io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:724)
        io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:650)
        io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:562)
        io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:997)
        io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
        io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
        java.base/java.lang.Thread.run(Thread.java:840)
```

xpromache, 2025-02-04:
Sorry for the slow reply. Please take a copy of the rocksdb database first (maybe I will ask you later to run some tests on it) then run a parameter archive rebuild of the concerned time interval:

https://docs.yamcs.org/yamcs-cli/yamcs_parameter-archive/

0lionelzhang0, 2025-02-04:
Thanks, I have copied the database to my local machine for testing. 

Based on the timestamp in the error (1737842642437 = 2025-01-25T22:04:02Z) I tried rebuilding from  
`yamcs parameter-archive rebuild 2025-01-24T00:00:00Z 2025-01-30T23:59:00Z`  
 However, I still got the same error

I then tried doing a full purge `yamcs parameter-archive purge` then reran the rebuild command above, and this resolved the issue. Is there a way to selectively purge a range of the parameter-archive?

xpromache, 2025-02-04:
There is no purge for a time range, it's not so straightforward to make given the structure of the data.

The rebuild is supposed to overwrite the data in the given range but it seems there is a bug there. I will review the code to try to find it. 

Does your archive contain values for that parameter at that timestamp (+max 2 hours)?

0lionelzhang0, 2025-02-07:
Sounds good, thanks. I just checked again on the deployed Yamcs instance, the error timestamp is 1737842758796 = 2025-01-25T22:05:58.796000Z

and this screenshot shows the last received packets for this parameter (last gen time 2025-01-25T22:05:24Z)

[image]

xpromache, 2025-02-11:
Hi Lionel, I only now noticed that the segmentStart is an odd number, so it must come from databases before Yamcs 5.10.0 when the merge operator has been introduced. Starting with hat version, the segments start always at intervals multiple of 2^23 milliseconds. The databases created before were not automatically upgraded (unless you did a purge+rebuild).
One of the reason for introducing the merge operator is to avoid segments starting at random times because it makes it difficult to search for them.

Can you please confirm that the database is coming from before Yamcs 5.10.0?

0lionelzhang0, 2025-02-11:
Hi Nicolae, just confirmed we are on 5.10.9  
[image]

The last time our pom.xml was modified was about a month ago from 5.10.6 -> 5.10.9, and before that was 3 months ago 5.10.5.

Error on our prod environment 1737842758796: 
[image]

The 1737842642437 came from when I copied the data to my local machine, I'm not sure why it's different (and odd number).

xpromache, 2025-03-04:
Can you please update to Yamcs 5.10.10 (or 5.11.4) and tell us if you still encounter problems? Best is to do a purge/rebuild to make sure it's all good.

0lionelzhang0, 2025-03-04:
Sounds good, thank you. We will update to 5.10.11 and monitor. We have only encountered one additional instance of DatabaseCorruptionException on 5.10.9 since this reported instance.

0lionelzhang0, 2025-03-05:
I don't see 5.10.11 on the maven repo yet. Will this get released soon?  
https://mvnrepository.com/artifact/org.yamcs/yamcs-core

xpromache, 2025-03-05:
Yesterday was my lucky day - I discovered another parameter archive corruption bug related to the realtime filler. If you are using that (you do if you started from the quickstart project), please update to the version 5.10.12.

It seems I forgot to release the 5.10.11 on maven; I just did it now together with 5.10.12
Still note that the maven website takes a while to update but the packages are already in their repository and should work if you change the pom.xml.