yamcs/yamcs #1009: Display aggregate parameters in a Procedure Stack

jdiez17, 2025-02-21:
[image]

Aggregate parameters are currently displayed as ellipsized text.

xpromache, 2025-02-21:
Not necessary the answer to your question but you could arrange for that command that the parameter shown there is recorded as the command response (or result). The complete and failed verifiers have a ReturnParamRef which can be used for this purpose.

jdiez17, 2025-02-21:
It sounds like a good idea to set ReturnParamRef for certain commands when it is known. But if I understand it correctly I'd have to replace my [default CompleteVerifier](https://gitlab.com/rccn/yamcs-instance/-/blob/master/src/main/yamcs/mdb/xtce-pus-frame-attempt2.xml?ref_type=heads#L311) (which checks for the PUS completion packet) for that command, i.e. I can't just add ReturnParamRef.

xpromache, 2025-02-21:
I see in the XTCE spec that you can have multiple CompleteVerifier, I don't know if it works in Yamcs though, you can try:

```xml
<complexType name="VerifierSetType">
                <sequence>
                        <element name="TransferredToRangeVerifier" type="xtce:TransferredToRangeVerifierType" minOccurs="0"/>
                        <element name="SentFromRangeVerifier" type="xtce:SentFromRangeVerifierType" minOccurs="0"/>
                        <element name="ReceivedVerifier" type="xtce:ReceivedVerifierType" minOccurs="0"/>
                        <element name="AcceptedVerifier" type="xtce:AcceptedVerifierType" minOccurs="0"/>
                        <element name="QueuedVerifier" type="xtce:QueuedVerifierType" minOccurs="0"/>
                        <element name="ExecutionVerifier" type="xtce:ExecutionVerifierType" minOccurs="0" maxOccurs="unbounded"/>
                        <element name="CompleteVerifier" type="xtce:CompleteVerifierType" minOccurs="0" maxOccurs="unbounded"/>
                        <element name="FailedVerifier" type="xtce:FailedVerifierType" minOccurs="0"/>
                </sequence>
</complexType>
```

fqqb, 2025-03-19:
Fix included in v5.11.7

closed 2025-03-19 by fqqb