yamcs/yamcs #1018: Are the ChangeAlarmRanges alarm types supported?

brownzach125, 2025-04-15:
I would like to use the ChangeAlarmRanges alarm types.

brownzach125, 2025-04-15:
I've tried adding them to my XTCE, but it didn't work. I've poked around the code base and documentation for clarity on if the feature is supported or not without any success.

fqqb, 2025-04-15:
It is not. Closest alternative I can think of, is to use an algorithm where you perform such logic, then have an enumerated output parameter (for example: "OK", "ERROR"), and then attach an EnumerationAlarm to that output parameter.

brownzach125, 2025-04-15:
Thanks for the suggestion. I've been trying to use algorithms following your suggestion. Does yamcs support InputParameterInstanceRef:instance?
I've been trying to compare a parameter sample to the previous instance of the parameter, but the values are always the same right now despite setting instance.

brownzach125, 2025-04-15:
[image]
Code snippet of the algorithm I'm trying to write. Ignore that I'm using the validCommandCount for testing purposes.

fqqb, 2025-04-16:
Yes, that looks fine. Alternatively you can also rely on the fact that algorithms keep their scope between multiple runs. So if you have a global variable in there, it could remember previous values.

brownzach125, 2025-04-16:
How do I compare telemetered values to local values? Something is not quite right here.

[image]

[image]

fqqb, 2025-04-16:
`rpm_feedback` is an object with multiple properties, one of which is the engineering value.
The engineering value should be accessible as `rpm_feedback.value`.

(same comment for your previous screenshot)

See: https://github.com/yamcs/yamcs/blob/yamcs-5.11.8/yamcs-core/src/main/java/org/yamcs/algorithms/ScriptAlgorithmExecutor.java#L313-L336

brownzach125, 2025-04-16:
Thanks for the help! I got confused by it working in my first snippet. I guess I just got unlucky.