yamcs/yamcs #1037: Stacks Editor not visible in Web UI when using authentication

Warp11, 2025-08-27:
I'm using the YamlAuthModule to assign user privileges. When using any role config except `superuser: true`, the Procedures->Stacks UI option of the web interface is always hidden. The page is still accessible and usable by entering the direct link to it, but the menu option is missing. I have not been able to find a config option related to this. 

Is this a bug or is there some config I have overlooked?

Thank you so much for your support and the great system you have developed!

fqqb, 2025-08-28:
Try adding `ReadBucket: ["stacks"]`

I see you have `ManageAnyBucket`, but it looks like that does not currently work (which is a bug to be fixed)

Warp11, 2025-08-28:
That did the trick, thank you so much for your quick response!

closed 2025-08-28 by Warp11